stephane-caron/lipm_walking_controller #2: Question about mc_tasks

junhyeokahn, 2018-09-21:
Hello,
Always appreciate your effort to maintain the open source!

I have a question about ```mc_tasks``` library. It seems to include ```CoMTask.h```, ```CoPTask.h```, ```OrientationTask.h``` and ```PostureTask.h``` but I could not find those files on 
[jrl-umi3218/Tasks](https://github.com/jrl-umi3218/Tasks) Repo. 
Would I request this dependancies to  __Pierre Gergondet__?

stephane-caron, 2018-09-25:
That's right, `mc_tasks` is part of `mc_rtc`. Thanks for pointing this out :) I just updated the README.

Actually `mc_tasks` is only a thin wrapper and you can find the implementation of all-but-one of these tasks in the [Tasks](https://github.com/jrl-umi3218/Tasks) repository:

- `CoMTask` is defined at [Tasks.h:411](https://github.com/jrl-umi3218/Tasks/blob/cc06abaa1d03394c347ee9f28515e50cf5cdf90c/src/Tasks/Tasks.h#L411)
- `OrientationTask` is defined at [Tasks.h:86](https://github.com/jrl-umi3218/Tasks/blob/cc06abaa1d03394c347ee9f28515e50cf5cdf90c/src/Tasks/Tasks.h#L86)
- `PostureTask` is defined at [Tasks.h:386](https://github.com/jrl-umi3218/Tasks/blob/cc06abaa1d03394c347ee9f28515e50cf5cdf90c/src/Tasks/Tasks.h#L386)

The only one that is not directly defined in _Tasks_ is the `CoPTask`, as it takes additional sensor readings. The CoP task is a [SurfaceTransformTask](https://github.com/jrl-umi3218/Tasks/blob/cc06abaa1d03394c347ee9f28515e50cf5cdf90c/src/Tasks/Tasks.h#L157) with feedback from force-torque sensors as described in Section III.A of [this document](https://hal.archives-ouvertes.fr/hal-01875387/document).

stephane-caron, 2018-10-29:
For some more info on this: all tasks in this inverse kinematics go to the cost function of the QP, while DOF position/velocity limits are inequality constraints.

For technical details on what's implemented in _Tasks_, you can take a look at the following paper: [Multi-robot and task-space force control with quadratic programming](https://hal.archives-ouvertes.fr/hal-01495662/).