stephane-caron/lipm_walking_controller #3: Questions about IK

andreadelprete, 2018-10-29:
Hi Stephane, 

thanks for sharing your code. Yesterday I've read your paper and I have a few questions about it.

The "foot damping control" and the "foot force difference control" compute desired foot velocities `v^c`(roll-pitch and z translation). I didn't get how these desired velocities are used in the acceleration-level IK introduced in Section III-D. 

- Is `v^c` used in the "maintain foot contacts" task? I don't see any other option, but this was not clear for me in the paper.
- When computing the acceleration-based tracking law (24) for this task, do you use the desired foot velocities `v^c` as `\dot{x}_c`? 
- Do you also integrate `v^c` and use it as `x_c`?
- What about `\ddot{x}_c` instead? Is it set to zero?

Other random remarks\questions:

- Once you compute the desired joint accelerations with IK, how do you translate them in motor commands? I guess you integrate them twice and send the resulting joint angles to the high-gain position control.
- In (18-21) some terms have not been defined
- I would reformulate (16) in a more clear form. It took me 5 minutes to figure out it was simply a proportional control over the CoP tracking error.
- Notation in (16) is not coherent with (18-23)

Thanks in advance for your answers.

stephane-caron, 2018-10-29:
Hi Andrea,

Thank you for your feedback :)

### Foot force control

> I didn't get how these desired velocities are used in the acceleration-level IK introduced in Section III-D.

All ``.^c`` quantities are directly fed as targets to the acceleration-level IK by Equation (24). In particular, ``v^c`` for a foot in contact is fed to ``\dot{x}^c`` in (24) as:
```
\dot{x}^c = {
   linear:  [0 0 v^c_z], 
   angular: [\dot{theta}^c_r \dot{theta}^c_p 0]
}
```
(Also, in (16) the \dot{theta} are actually \dot{theta}^c.) I will update the manuscript to make this clearer.

> Do you also integrate v^c and use it as x_c?

In the reported experiments we didn't add integration to ``x^c``. It is implemented in mc_rtc in the next version of the ``AdmittanceTask`` (the one described in [Fig. 4 of this paper](https://hal.archives-ouvertes.fr/hal-01495662/document)) but in these experiments we kept ``x^c`` at the desired contact frame. It matters mildly here because contact tasks have a stiffness significantly lower than their damping (``K=1`` versus ``B=300``).

> What about \ddot{x}_c instead? Is it set to zero?

Yes, in the reported experiments we just set ``\ddot{x}^c = 0``, while in the next version of the task ([Fig. 4 of this paper](https://hal.archives-ouvertes.fr/hal-01495662/document)) ``\ddot{x}^c`` is taken proportional to the filtered time derivative of the measured contact wrench.

### Other remarks/questions

> Once you compute the desired joint accelerations with IK, how do you translate them in motor commands? I guess you integrate them twice and send the resulting joint angles to the high-gain position control.

Yes.

> In (18-21) some terms have not been defined

I'll add the missing definitions of ``A_dfz`` and ``T_vdc``. Let me know if I forgot some more.

> I would reformulate (16) in a more clear form. It took me 5 minutes to figure out it was simply a proportional control over the CoP tracking error.

Note that there is a difference here: straightforward proportional control over the CoP tracking error ``A * (p^d - p^m)`` rewrites to ``A' (p^d x f^m - tau^m)`` where ``A'`` depends on the measured contact force:
```
     A [n]_\cross
A' = ------------
        n . f^m
```
(``n`` is the normal of the contact frame.) In particular this ``A'`` may diverge when pressure drops to zero. With (16) this ``A'`` is taken as a constant. (You can also characterize this task equivalently as proportional ankle torque tracking with a measure-dependent torque target ``tau^d = p^d x f^m``.)

Thanks again for your feedback. I will update the report accordingly, and hope these answers help clarify!

andreadelprete, 2018-10-29:
Thanks for the answers. Everything's clear now!
Cheers,
Andrea

closed 2018-10-30 by stephane-caron