stephane-caron/lipm_walking_controller #4: Error while strting the package: undefined symbol: LOAD_GLOBAL

mohamadmahdavian, 2019-06-25:
Hello Dr.Caron

I have installed and imported my robot model in this package. But I get following error. Do you have any idea what the problem is?

################################################
Loading additional global configuration etc/mc_rtc.conf
GUI server enabled
Will serve data on:
- ipc:///tmp/mc_rtc_pub.ipc
- tcp://*:4242
Will handle requests on:
- ipc:///tmp/mc_rtc_rep.ipc
- tcp://*:4343
Loading additional controller configuration"/usr/local/lib/mc_controller//etc/LIPMWalking.conf"
Loading additional controller configuration"/home/saeed/.config/mc_rtc/controllers/LIPMWalking.conf"
Create controller LIPMWalking
MCController(base) ready
Attempt to open /usr/local/lib/mc_controller/lipm_walking_controller/states/DoubleSupport.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/lipm_walking_controller/states/DoubleSupport.so
/usr/local/lib/mc_controller/lipm_walking_controller/states/DoubleSupport.so: undefined symbol: LOAD_GLOBAL
New state from library: DoubleSupport
Attempt to open /usr/local/lib/mc_controller/lipm_walking_controller/states/SingleSupport.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/lipm_walking_controller/states/SingleSupport.so
/usr/local/lib/mc_controller/lipm_walking_controller/states/SingleSupport.so: undefined symbol: LOAD_GLOBAL
New state from library: SingleSupport
Attempt to open /usr/local/lib/mc_controller/lipm_walking_controller/states/Standing.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/lipm_walking_controller/states/Standing.so
/usr/local/lib/mc_controller/lipm_walking_controller/states/Standing.so: undefined symbol: LOAD_GLOBAL
New state from library: Standing
Attempt to open /usr/local/lib/mc_controller/lipm_walking_controller/states/Initial.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/lipm_walking_controller/states/Initial.so
/usr/local/lib/mc_controller/lipm_walking_controller/states/Initial.so: undefined symbol: LOAD_GLOBAL
New state from library: Initial
Robots loaded in FSM controller:
- jvrc-1
- ground
Added task FSM_posture_jvrc-1
Initialized new ModelPredictiveControl solver
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted (core dumped)
#############################################

Thank You
Regards

stephane-caron, 2019-06-26:
Hmm this looks like an mc_rtc issue. @gergondet any idea what could be going on here?

I don't have the `LOAD_GLOBAL` symbol in my FSM shared objects either (e.g. ``nm -g DoubleSupport.so``) and it's not a problem for me when running mc_vrep.

mohamadmahdavian, 2019-06-27:
Hello Dr.Caron

Thank you for the follow up.Actually, I have another question regarding the same issue. 
When I am trying to build the current lipm_walking package with the latest mc_rtc package I get several errors including(as an example) setcriticalgains not found which I believe is because of version differences between two packages. So could you please update your code based on the latest mc_rtc code or can you give me the commit number of the mc_rtc that you are using?(I think my previous question is comming from the same issue)

Thank you

stephane-caron, 2019-06-27:
Sure, you have a set of working commits for all dependencies in [COMMITS.md](https://github.com/stephane-caron/lipm_walking_controller/blob/master/COMMITS.md).

mohamadmahdavian, 2019-07-02:
Thank you for the respond.I compilled all projects with proper commits you provided and they worked out well till compiling mc_rtc_ros where I get:

-- +++ processing catkin package: 'mc_rtc_ticker'
-- ==> add_subdirectory(mc_rtc_ros/mc_rtc_ticker)
-- mc_rtc is required.
-- Checking for module 'mc_rtc'
--   No package 'mc_rtc' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  mc_rtc_ros/mc_rtc_ticker/cmake/pkg-config.cmake:290 (PKG_CHECK_MODULES)
  mc_rtc_ros/mc_rtc_ticker/cmake/pkg-config.cmake:502 (ADD_DEPENDENCY)
  mc_rtc_ros/mc_rtc_ticker/CMakeLists.txt:9 (add_required_dependency)

-- Configuring incomplete, errors occurred!
See also "/home/mohammad/catkin_ws/lipm/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/mohammad/catkin_ws/lipm/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Do you have any Idea?
Thank You

stephane-caron, 2019-07-05:
Well it looks like you don't have mc_rtc installed on your system. Could you successfully ``make`` and ``sudo make install`` in your mc_rtc build folder?

mohamadmahdavian, 2019-07-06:
The first error that I get is when I am using cmake .. command and the error is:
########

-- Checking for module 'eigen3 >= 3.0.0'
--   Found eigen3 , version 3.3.7
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   timer
--   regex
--   chrono
--   atomic
-- Tasks is required.
-- Checking for module 'Tasks'
--   Found Tasks, version UNKNOWN-dirty
-- Tasks_d is optional
-- Checking for module 'Tasks_d'
--   No package 'Tasks_d' found
-- Pkg-config module Tasks vUNKNOWN-dirty has been detected with success.
-- mc_rbdyn_urdf is required.
-- Checking for module 'mc_rbdyn_urdf'
--   Found mc_rbdyn_urdf, version UNKNOWN-dirty
-- mc_rbdyn_urdf_d is optional
-- Checking for module 'mc_rbdyn_urdf_d'
--   No package 'mc_rbdyn_urdf_d' found
-- Pkg-config module mc_rbdyn_urdf vUNKNOWN-dirty has been detected with success.
-- hrpsys-base is optional.
-- Checking for module 'hrpsys-base'
--   No package 'hrpsys-base' found
-- nanomsg is optional.
-- Checking for module 'nanomsg'
--   Found nanomsg, version 1.1.5-3-g53d13b3
-- Pkg-config module nanomsg v1.1.5-3-g53d13b3 has been detected with success.
-- Found mc_env_description: /usr/local/share/mc_env_description
-- roscpp is optional.
-- Checking for module 'roscpp'
--   Found roscpp, version 1.12.14
-- Pkg-config module roscpp v1.12.14 has been detected with success.
-- nav_msgs is required.
-- Checking for module 'nav_msgs'
--   Found nav_msgs, version 1.12.7
-- Pkg-config module nav_msgs v1.12.7 has been detected with success.
-- sensor_msgs is required.
-- Checking for module 'sensor_msgs'
--   Found sensor_msgs, version 1.12.7
-- Pkg-config module sensor_msgs v1.12.7 has been detected with success.
-- tf2_ros is required.
-- Checking for module 'tf2_ros'
--   Found tf2_ros, version 0.5.20
-- Pkg-config module tf2_ros v0.5.20 has been detected with success.
-- rosbag is required.
-- Checking for module 'rosbag'
--   Found rosbag, version 1.12.14
-- Pkg-config module rosbag v1.12.14 has been detected with success.
-- mc_rtc_msgs is required.
-- Checking for module 'mc_rtc_msgs'
--   Found mc_rtc_msgs, version 0.1.0
-- Pkg-config module mc_rtc_msgs v0.1.0 has been detected with success.
-- Looking for C++ include ltdl.h
-- Looking for C++ include ltdl.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Looking for stdint.h
[... 48 lines cut ...]
    linked by target "mc_tasks" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src
    linked by target "mc_control" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src
    linked by target "int_obj" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_robots
    linked by target "json" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_robots
    linked by target "env" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_robots
    linked by target "jvrc-1" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_robots
    linked by target "halfsitpose" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control
    linked by target "posture" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/samples/Posture
    linked by target "body6d" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/samples/Body6d
    linked by target "com" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/samples/CoM
    linked by target "text" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/samples/Text
    linked by target "fsm" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/fsm
    linked by target "mc_control_fsm" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/fsm
    linked by target "python" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/src/mc_control/python
    linked by target "test_controller_ticker" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/tests
    linked by target "testQuadraticGenerator" in directory /home/mohammad/catkin_ws/lipm/mc_rtc/tests

-- Configuring incomplete, errors occurred!
See also "/home/mohammad/catkin_ws/lipm/mc_rtc/build/CMakeFiles/CMakeOutput.log".
See also "/home/mohammad/catkin_ws/lipm/mc_rtc/build/CMakeFiles/CMakeError.log".
####

Also another issue is that the git commit of "copra" package of the COMMITS.md file can not be founded.

Thank You

stephane-caron, 2019-07-08:
The first error being related to mc_rtc, I suggest you post it on the [mc_rtc issue tracker](https://gite.lirmm.fr/multi-contact/mc_rtc/issues).

Regarding _copra_, you can use the latest commit from its master branch (``692f4a83...``). I just checked that it compiles and runs fine with this controller.

closed 2019-07-08 by stephane-caron