stephane-caron/lipm_walking_controller #5: Dynamic Simulation

Ahmadreza-Shahrokhshahi, 2019-07-05:
Hi Dr. Caron,
I have studied your paper which is corresponding to the LIPM walking controller package. I want to ask you about the mc-rtc package which is linked with RViz; I just wanted to make sure if it is a dynamic simulation environment? 
If so, why did you use Choreonoid?
How did you see these two environments, and how you think these two are different / in which aspects?

Thank you

stephane-caron, 2019-07-05:
Actually mc_rtc is a robot control library, which is different from a forward dynamics simulator. Here is how these two different components interact:
[image]
The role of Choreonoid is to simulate what our controller will see (sensor measurements) when running on the actual robot. It is one of the important tools in the testing pipeline. Once the behavior of the controller has been validated in simulation, we can run it on the real robot.

closed 2019-07-05 by stephane-caron

mohamadmahdavian, 2019-07-05:
Hi Dr.Caron

Thank you for your response. In your paper you have talked about the advantages of Choreonoid and yet you have used rviz and VREP in this package. I wanted to know your idea about these environments. We need to have a simulation environment with GRF/M feedback and I wanted to know your idea about the best one. Also, if your answer is Choreonoid, I would like to know if we need to make any adjustments in the installation process. So we would have our robot in choreonoid and connect mc-rtc package to it. 

Thank You
Regards

stephane-caron, 2019-07-08:
There are couple of points to note here:

- RViz is a visualization application, not a dynamics simulator. We use RViz to see what's going on on the robot whether running in dynamic simulation or on robot hardware.
- Choreonoid is a complete robotics environments, but we only [use it as a dynamics simulator](https://choreonoid.org/en/about.html#using-choreonoid-as-a-dynamics-simulator). (Choreonoid relies on the AIST physics engine.) 
- V-REP is also a complete robotics environment, and we also [use it only as a dynamics simulator](http://www.coppeliarobotics.com/helpFiles/en/dynamicsModule.htm). It offers three physics engines: Bullet, ODE and Newton. (Newton is the best of these three.)

As discussed in [section VI.B of this paper](https://hal.archives-ouvertes.fr/hal-01689331/document), Choreonoid with the AIST physics engine is the best simulator in terms of real-timeness and plausibility of contact forces. The installation process is however a bit more complicated than for V-REP. For more details, check out the related mc interfaces:

- [mc_openrtm](https://gite.lirmm.fr/multi-contact/mc_openrtm) to use mc_rtc with Choreonoid
- [mc_vrep](https://gite.lirmm.fr/multi-contact/mc_vrep) to use mc_rtc with V-REP

stephane-caron, 2019-07-08:
PS: to illustrate what I mean by "plausibility" of contact forces, here is a sample run of the lipm_walking_controller in V-REP with Newton as physics engine:
- [walk_in_vrep.mp4.zip](https://github.com/stephane-caron/lipm_walking_controller/files/3368628/walk_in_vrep.mp4.zip)

The robot runs the ``straight_20cm_steps`` plan, hitting and pushing a 2 kg box on the way for disturbance. We can see a jerkiness in contact forces that is not realistic (F/T measurements on the real robot are much smoother!)