stephane-caron/lipm_walking_controller #6: LIPM package with Choreonoid

mohamadmahdavian, 2019-07-17:
Hi Dr.Caron

I have installed all dependencies and also installed mc_rtc successfully with cmake ../make/make install. I couldn't install mc_rtc  with ./build_and_install.sh ,because I wanted to use choreonoid instead of V-REP. After that I installed hrpsys and mc_openRTM to use with choreonoid. In the next step I installed Choreonoid and now I want  to import my robot model into choreonoid and connect the mc_rtc to it. But it is not included in the tutorial. First, I don't know how should I import my robot in Choreonoid. I can see some packages like "HRP4J" in mc_openrtm with .cnoid files. Should I make the same files for my robot? If yes, Is there any generator for them and if the answer is no how should I import my robot in choreonoid?
Also, I would like to know how should I connect choreonoid to mc_rtc. There is a plugin for choreonoid which connects choreonoid to ros. Should I use that one or there is another way?
 
Thank You
Regards

stephane-caron, 2019-07-17:
OK great. Note that your questions are actually not related to the LIPM walking controller, rather to its dependencies. I will give you some first pointers here, and direct you to the correct issue tracker to continue the discussion.

### How to import your robot model in Choreonoid?

For this, you will need to convert your robot model to VRML. There is documentation for this process in the [OpenHRP user's manual](https://fkanehiro.github.io/openhrp3-doc/en/users_manual.html) such as:

- [Overview of the VRML format](https://fkanehiro.github.io/openhrp3-doc/en/create_model.html)
- [Sample humanoid model in VRML](https://fkanehiro.github.io/openhrp3-doc/en/sample_model.html)

There are also some scripts to automate the conversion from and to VRML in the [simtrans](https://github.com/fkanehiro/simtrans) repository.

### How to connect mc_rtc and Choreonoid

The good place to discuss this is in the issue tracker of the [mc_openrtm](https://gite.lirmm.fr/multi-contact/mc_openrtm) project. Let's continue the discussion there, where @gergondet and other competent people can give you feedback.

closed 2019-07-17 by stephane-caron

mohamadmahdavian, 2019-07-17:
Thank you Dr.Caron for your respond.