stephane-caron/lipm_walking_controller #12: ZMP constraint in MPC during stair climbing

stephane-caron, 2019-10-06:
Moving @Saeed-Mansouri's second question from #9 to this separate issue:

> As discussed in “Walking Trajectory Planning on Stairs Using Virtual Slope for Biped Robots”, there are two major issues for motion generation of bipeds on stairs: i) the ZMP equation problem: which is “the ZMP equation on stairs is different from that of flat ground”, and ii) the ZMP definition problem: which is “the ZMP cannot be defined in the double-support phase on stairs because the contact points of the feet do not constitute a plane.” In this paper, a virtual slope method is proposed to resolve these issues. I cannot find any difference in the approach implemented in LIPM to generate straight and stair walking trajectories. Would you please explain how you handle these two issues in the LIPM for stair walking?

stephane-caron, 2019-10-06:
Regarding the definition of the ZMP: the set of [zero-tilting moment points (ZMPs)](https://scaron.info/teaching/zero-tilting-moment-point.html) of a wrench is actually an axis. For practical purposes we can intersect it with a plane, but note that other parameterizations of this axis such as the [eCMP or VRP](https://doi.org/10.1109/TRO.2015.2405592) do not make such intersection.

When walking on flat floor, taking this plane as the floor plane has practical advantages (for instance, the ZMP support area then becomes the convex hull of ground contact points, although this construction is not valid in general). In the general case, we can always [take the ZMP in any plane and construct its support area by polyhedral projection](https://hal.archives-ouvertes.fr/hal-02108589/document).

stephane-caron, 2019-10-06:
Regarding the ZMP constraint used in the MPC: the model predictive control QP that this controller uses for walking pattern generation follows the formulation from Pierre-Brice Wieber's group, see
e.g. [this paper](https://hal.inria.fr/hal-01418355/document). One important point in this formulation is that *the ZMP constraint is only added during single support phases* (SSP). In a typical use case:

- SSP duration is 0.7 s
- double support phase (DSP) duration is 0.1 s
- MPC timestep is 0.1 s

Also, the output ZMP trajectory is continuous and roughly piecewise linear. In the [version from the University of Rome](https://arxiv.org/pdf/1901.08505.pdf), which parameterizes trajectories in ZMP velocity rather than CoM jerk, it is exactly piecewise linear. Next, by convexity of the support area, a necessary and sufficient condition to ensure that the ZMP lies inside the area during DSP is that (1) it lies in the previous SSP area at the beginning of DSP and (2) it lies in the next SSP area at the end of DSP. Then, at MPC timesteps, you end up only checking SSP areas.

This controller uses this simplification as of 2a1220daa1f87556cc86fc2e3efc8de8c8b4f30b. It explains why there is no difference between horizontal walking and stair climbing in this MPC: as long as foot contacts are horizontal, they need not be coplanar and we don't have to decide in which plane to consider the ZMP during DSP.

On a side note, if foot contacts are not horizontal, no big deal: we can just rectify them using the analytical projection of their [contact wrench cone](https://scaron.info/teaching/wrench-friction-cones.html) described in [Section IV of this paper](https://hal.archives-ouvertes.fr/hal-01349880/document). I didn't implement this in this controller since non-horizontal foot contacts were not part of our use case.

closed 2019-10-18 by stephane-caron

Saeed-Mansouri, 2020-10-22:
I have found that the plane of the support foot is considered to calculate the zmp of the pendulum (`pendulum.zpm()`) for both SSP and DSP. So the zmp is estimated on different planes when the walking state is switched from DSP to SSP. I know that the set of zero-tilting moment points (ZMPs) of a wrench is an axis and for practical purposes, we can intersect it with a plane. But I couldn't understand how we can consider some different planes for walking path planning. Would you please clear it for me?