stephane-caron/lipm_walking_controller #20: Member setBaseLinkVelocity is missing

ndehio, 2019-10-22:
I updated to the newest mc_rtc version today (branch master) and am now facing errors when trying to recompile lipm_walking.

```
/home/idh-user/src/multi-contact/lipm_walking_controller/src/Controller.cpp: In member function ‘void lipm_walking::Controller::internalReset()’:
/home/idh-user/src/multi-contact/lipm_walking_controller/src/Controller.cpp:476:20: error: ‘struct mc_rbdyn::Robot’ has no member named ‘setBaseLinkVelocity’
     controlRobot().setBaseLinkVelocity(Eigen::Vector6d::Zero());
                    ^
/home/idh-user/src/multi-contact/lipm_walking_controller/src/Controller.cpp:478:17: error: ‘struct mc_rbdyn::Robot’ has no member named ‘setBaseLinkVelocity’
     realRobot().setBaseLinkVelocity(Eigen::Vector6d::Zero());
                 ^
```

There will be a demo in our lab tomorrow... ;-)

gergondet, 2019-10-22:
This has been replace with `mc_rbdyn::Robot::velW(const sva::ForceVecd &)`. The function is otherwise identical

stephane-caron, 2019-10-22:
Note: at a given time, the controller is guaranteed to compile with the related project commits listed in [COMMITS.md](https://github.com/stephane-caron/lipm_walking_controller/blob/master/COMMITS.md) (unfortunately we don't have a proper version system here). Use more recent versions at your own peril :wink:

ndehio, 2019-10-22:
It should be MotionVecd instead of ForceVecd. The following worked for me:
```
    controlRobot().velW(sva::MotionVecd(Eigen::Vector6d::Zero()));
    realRobot().posW(X_0_fb);
    realRobot().velW(sva::MotionVecd(Eigen::Vector6d::Zero()));
```
Thanks for your help

ndehio, 2019-10-22:
@gergondet are you sure that these functions are identical?

When clicking on "start standing" in the FSM the robot just falls backward in choreonoid... It seems there are mc_rtc changes (not necessarily in the velW function) that influence the controller.

ndehio, 2019-10-23:
@gergondet fixed the problem yesterday:
In src/FloatingBaseObserver.cpp you have to modify:
```
    //Eigen::Matrix3d E_0_mIMU = realRobot.bodySensor().orientation().toRotationMatrix();
    Eigen::Matrix3d E_0_mIMU = controlRobot_.bodySensor().orientation().toRotationMatrix();
```
Also, in the config file add
```
"RunObservers": ["Encoder"],
"UpdateObservers": ["Encoder"],
```
works for me =)

arntanguy, 2019-10-23:
We had that discussion yesterday with Pierre. The way I see it: 

- `robot()` should always contain the raw measurements passed to `MCGlobalController`. 
- `realRobot()` should contain estimated sensors (such as the floating base, filtered forces, etc). It'd be reasonable to default to setting the `realRobot()` sensors to the same value as `robot()`.

Indeed after the observers MR we are no longer setting `realRobot(). bodySensor ()`. I originally intended the observers to do that, but in the end I didn't, which results in an illogical situation.

Feel free to open a merge request to restore the former behaviour (I'd do it myself, but I'm on my phone). Basically in the setters of `MCGlobalController` you can set the sensor values for both robots by default.

The observers would then be free to modify this sensor reading if they wish.

@Niels if you're not using the floating base observer, I don't see how the fix you mention would have any effect (also in mc_rtc matter this is now `src/mc_observers/KinematicInertialObserver.cpp`)

arntanguy, 2019-10-23:
Oh my bad, it's Stéphane's floating base observer from this repository we're taking about. Then yeah that'll do.

I hope I can continue on integrating this and cleaning up this controller to avoid duplication with mc_rtc as soon as I'm in AIST ;)

closed 2019-10-24 by stephane-caron