stephane-caron/lipm_walking_controller #21: "no controller selected" error

mohamadmahdavian, 2019-10-23:
Hi Dr. Caron,

I have fixed mc_rtc, mc_vrep, mc_rbdyn issues that I had with my custom robot and I am finally able to run lipm controller. Unfortunately, I still get error by running mc_vrep:

> GUI server enabled
> Will serve data on:
> - ipc:///tmp/mc_rtc_pub.ipc
> - tcp://*:4242
> Will handle requests on:
> - ipc:///tmp/mc_rtc_rep.ipc
> - tcp://*:4343
> Loading additional controller configuration"/usr/local/lib/mc_controller//etc/LIPMWalking.conf"
> Loading additional controller configuration"/home/mohammad/.config/mc_rtc/controllers/LIPMWalking.conf"
> Controller LIPMWalking enabled in configuration but not available
> Create controller HalfSitPose
> MCController(base) ready
> Added task posture_beta1_v4
> MCHalfSitPoseController init done 0x18769b0
> No controller selected or selected controller is not enabled, please check your configuration file
> No controller enabled
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  No controller enabled
> Aborted (core dumped)

Regards

stephane-caron, 2019-10-23:
From the error messages:

> Controller LIPMWalking enabled in configuration but not available

This means your configuration asks for the "LIPMWalking" controller, but mc_rtc failed to load its dynamic library. Usually it's a linking problem. Check for any error line in:
```
$ cd /usr/local/lib/mc_controller
$ ldd liblipm_walking_controller.so
```
This is a common error for mc_rtc developers. @gergondet can you point us to some documentation on how to deal with it?

mohamadmahdavian, 2019-10-23:
The file is there along with other controllers. ldd shows no error, all libraries are linked properly. could it be the version mismatch between lipm controller, mc_rtc and other components?

gergondet, 2019-10-28:
Hi @mohamadmahdavian,

You can try to run:

```bash
LD_DEBUG=libs mc_vrep 2> ld.log
```

Then you can check ld.log for more specific linking issues. In the file you search for when the linker is attempting to load LIPMWalking.so

If you are building with catkin, don't forget to run `catkin_make install` if you want it to work without altering your configuration.

mohamadmahdavian, 2019-10-28:
Thanks @gergondet for this informative feedback. I ran the command and had a look into the file:
1. almost all packages are loaded properly: tasks, mc_rtc_ros, rbdyn, etc.
2. a lot of objects including my robot definition mc_robots/beta1_v4.so has the following error:
>  error: symbol lookup error: undefined symbol: beta1_v4_LTX_MC_RTC_ROBOT_MODULE (fatal)
>  error: symbol lookup error: undefined symbol: beta1_v4_LTX_LOAD_GLOBAL (fatal)
>  error: symbol lookup error: undefined symbol: LOAD_GLOBAL (fatal)
3. lipmwalking.so (@stephane-caron) also has the following error:
>  error: symbol lookup error: undefined symbol: _ZN12lipm_walking10Stabilizer14MAX_FDC_RZ_VELE (fatal)

I have attached the file itself for further details.
[ld.log](https://github.com/stephane-caron/lipm_walking_controller/files/3780704/ld.log)

Regards,

gergondet, 2019-11-04:
```
error: symbol lookup error: undefined symbol: _ZN12lipm_walking10Stabilizer14MAX_FDC_RZ_VELE (fatal)
```

That's the root error actually, this happens with recent versions of GCC... I will submit a clean patch for it

stephane-caron, 2019-11-04:
Thanks @gergondet! I pushed your fix in bff8c03d64118ad3e3cc40585ab72d9877776e0e. Can you guys confirm that it fixes the problem with your versions of GCC?

gergondet, 2019-11-08:
@stephane-caron I confirmed on slack but I'm confirming it's working with your fix I hope this solves @mohamadmahdavian problem as well

mohamadmahdavian, 2019-11-09:
Hi Dr. @stephane-caron  and Dr. @gergondet 

I installed the entire stack from scratch based on COMMITS.md file, and this time I tried JVRC-1 (jvrc_description) robot instead of my custom robot and any other changes to the system. Besides all the installation difficulties, this time I ran into this error:

```

Warning: ${MODEL} of "${MODEL}/JVRC1/main.wrl" is not defined.
MODEL of "${MODEL}/JVRC1/main.wrl" cannot be expanded.
Error: "JVRC1" cannot be restored.
Restoring BodyItem "longfloor"
Warning: ${MODEL} of "${MODEL}/longfloor.wrl" is not defined.
MODEL of "${MODEL}/longfloor.wrl" cannot be expanded.
Error: "longfloor" cannot be restored.
Restoring AISTSimulatorItem "AISTSimulator"
Restoring PythonSimScriptItem "sim_mc_udp.py"
Loading Python Script for Simulation "/usr/local/share/hrpsys/samples/JVRC1/sim_mc_udp.py"
 -> ok!
4 / 6 item(s) have been loaded.
Warning: 2 item(s) were not loaded.
Project "sim_mc_udp.cnoid" has been partially loaded.
Simulation by AISTSimulator has started.
Execution of Python script "sim_mc_udp.py" has been started.
[rtm.py] nshost already set as localhost
[rtm.py] nsport already set as 2809
[rtm.py] Failed to parse corba.master_manager, use 2810
[rtm.py] configuration ORB with localhost:2809
[rtm.py] configuration RTCManager with localhost:2810
[rtm.py] trying to findRTCManager on port2810
The execution of Python script "sim_mc_udp.py" failed.
AttributeError: 'NoneType' object has no attribute 'port'

At:
  /usr/local/share/hrpsys/samples/JVRC1/sim_mc_udp.py(113): init
  /usr/local/share/hrpsys/samples/JVRC1/sim_mc_udp.py(152): <module>
```

- When I change the cnoid file and address the model directly to the main.wrl file, it crashes.
* I believe it is the best to have a docker file (or some other way) to install the stack in a separate container from scratch. Currently, you have an environment which any part of it could be different than installing it from scratch.

Thank You
Best Regards

stephane-caron, 2019-11-11:
@mohamadmahdavian this new problem is related to the JVRC1 Choreonoid project, so I'm moving the issue to [multi-contact/jvrc_choreonoid](https://gite.lirmm.fr/multi-contact/jvrc_choreonoid/issues). Let's continue the discussion there.

closed 2019-11-11 by stephane-caron

stephane-caron, 2019-12-10:
> I believe it is the best to have a docker file (or some other way) to install the stack in a separate container from scratch.

@mohamadmahdavian I prepared a [Docker image](https://hub.docker.com/r/stephanecaron/lipm_walking_controller) for the release of [v1.5](https://github.com/stephane-caron/lipm_walking_controller/releases/tag/v1.5) of the controller. You can try it out directly by:

```
xhost +local:docker
docker run -it --rm --user ayumi \
    -e DISPLAY=${DISPLAY} -v /tmp/.X11-unix:/tmp/.X11-unix:rw \
    stephanecaron/lipm_walking_controller \
    lipm_walking --floor
```