stephane-caron/lipm_walking_controller #23: Missing hrp4_description

rshum19, 2019-10-24:
Hi Stephane,

I've read your paper and find it really interesting. Thanks for sharing your code and work. 

I'm having trouble running the example in the README. When I try to launch 
```
roslaunch lipm_walking_controller display.launch robot:=hrp4
```
I get the error: 
```
ResourceNotFound: hrp4_description
```
I'm assuming this because I am missing  *hrp4_description* package. However, I don't seem to be able to find it anywhere. Could you help me pointing me to it?

Thank you,
Roberto

stephane-caron, 2019-10-24:
Hi @rshum19, thanks! Unfortunately, the `hrp4_description` package contains the URDF model of the HRP-4 robot, which we are not allowed to share beyond our team as it is intellectual property of Kawada Industries.

Let's leave this issue open until I add [JVRC-1](https://github.com/jvrc/model) to the list of available robot models. This one is publicly released.

rshum19, 2019-10-24:
Hi @stephane-caron,

That is unfortunate. I look forward to when the JVRC-1 is available and thank you for making all the control code open source!

stephane-caron, 2019-11-02:
Almost there! PR https://github.com/stephane-caron/lipm_walking_controller/pull/25 adds JVRC-1 to the list of available robot models in the controller configuration.

I have prepared a Choreonoid project to use JVRC-1 with mc_rtc. We will release it after testing it in the team next week. I will merge the PR then.

stephane-caron, 2019-11-05:
@rshum19 the JVRC-1 model is now available :) You will find the VRML model and Choreonoid project files to use it with mc_rtc in [multi-contact/jvrc_choreonoid](https://gite.lirmm.fr/multi-contact/jvrc_choreonoid).

closed 2019-11-05 by stephane-caron

rshum19, 2019-11-05:
@stephane-caron Thank you for letting me know and getting the JVRC-1 model integrated so promptly! Looking forward to testing it out! 

Again thank you!