stephane-caron/lipm_walking_controller #29: Initial state is not valid

ndehio, 2019-11-14:
I just upgraded to the recent version (master branch).
When starting I receive this error message:

```
MCUDPControl -h localhost -f ~/.config/mc_rtc/controllers/LIPMWalking.conf
Loading additional global configuration "/home/idh-user/.config/mc_rtc/mc_rtc.conf"
Loading additional global configuration /home/idh-user/.config/mc_rtc/controllers/LIPMWalking.conf
GUI server enabled
Will serve data on:
- ipc:///tmp/mc_rtc_pub.ipc
- tcp://*:4242
Will handle requests on:
- ipc:///tmp/mc_rtc_rep.ipc
- tcp://*:4343
Loading additional controller configuration"/usr/local/lib/mc_controller//etc/FogaleController.conf"
Create controller FogaleController
MCController(base) ready
Attempt to open /usr/local/lib/mc_controller/fsm/states/PythonState.so
Found matching class name symbol MC_RTC_FSM_STATE
New state from library: Python
Attempt to open /usr/local/lib/mc_controller/fsm/states/Grippers.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/Grippers.so
/usr/local/lib/mc_controller/fsm/states/Grippers.so: undefined symbol: LOAD_GLOBAL
New state from library: Grippers
Attempt to open /usr/local/lib/mc_controller/fsm/states/SlidingFootContact.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/SlidingFootContact.so
/usr/local/lib/mc_controller/fsm/states/SlidingFootContact.so: undefined symbol: LOAD_GLOBAL
New state from library: SlidingFootContact
Attempt to open /usr/local/lib/mc_controller/fsm/states/Pause.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/Pause.so
/usr/local/lib/mc_controller/fsm/states/Pause.so: undefined symbol: LOAD_GLOBAL
New state from library: Pause
Attempt to open /usr/local/lib/mc_controller/fsm/states/Meta.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/Meta.so
/usr/local/lib/mc_controller/fsm/states/Meta.so: undefined symbol: LOAD_GLOBAL
New state from library: Meta
Attempt to open /usr/local/lib/mc_controller/fsm/states/HalfSitting.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/HalfSitting.so
/usr/local/lib/mc_controller/fsm/states/HalfSitting.so: undefined symbol: LOAD_GLOBAL
New state from library: HalfSitting
Attempt to open /usr/local/lib/mc_controller/fsm/states/AddRemoveContact.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/AddRemoveContact.so
/usr/local/lib/mc_controller/fsm/states/AddRemoveContact.so: undefined symbol: LOAD_GLOBAL
New state from library: AddRemoveContact
Attempt to open /usr/local/lib/mc_controller/fsm/states/Parallel.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/Parallel.so
/usr/local/lib/mc_controller/fsm/states/Parallel.so: undefined symbol: LOAD_GLOBAL
New state from library: Parallel
Attempt to open /usr/local/lib/mc_controller/fsm/states/MetaTasks.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fsm/states/MetaTasks.so
/usr/local/lib/mc_controller/fsm/states/MetaTasks.so: undefined symbol: LOAD_GLOBAL
New state from library: MetaTasks
Attempt to open /usr/local/lib/mc_controller/fogale-controller/states/Demo.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fogale-controller/states/Demo.so
/usr/local/lib/mc_controller/fogale-controller/states/Demo.so: undefined symbol: LOAD_GLOBAL
New state from library: Fogale::Demo
Attempt to open /usr/local/lib/mc_controller/fogale-controller/states/Calibrate.so
Found matching class name symbol MC_RTC_FSM_STATE
Could not get symbol LOAD_GLOBAL in library /usr/local/lib/mc_controller/fogale-controller/states/Calibrate.so
/usr/local/lib/mc_controller/fogale-controller/states/Calibrate.so: undefined symbol: LOAD_GLOBAL
New state from library: Fogale::Calibrate
Robots loaded in FSM controller:
- hrp4
- ground
Added task FSM_posture_hrp4
The initial state you provided (Initial) is not valid, will replace it with the first valid state in the transitions map: Pause
[1]    15833 segmentation fault (core dumped)  MCUDPControl -h localhost -f ~/.config/mc_rtc/controllers/LIPMWalking.conf
```

Is this `LOAD_GLOBAL` important? How is it possible that a state becomes invalid?

ndehio, 2019-11-14:
aha, there is an issue with copra
```
ldd -r -d /usr/local/lib/mc_controller/lipm_walking_controller.so 
	linux-vdso.so.1 =>  (0x00007ffe6864d000)
	liblipm_walking_controller.so => /usr/local/lib/liblipm_walking_controller.so (0x00007f422c1b6000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f422bfb2000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f422bc30000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f422ba1a000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f422b650000)
	libmc_control_fsm.so.1 => /usr/local/lib/libmc_control_fsm.so.1 (0x00007f422b3f0000)
	libmc_control.so.1 => /usr/local/lib/libmc_control.so.1 (0x00007f422b160000)
	libmc_tasks.so.1 => /usr/local/lib/libmc_tasks.so.1 (0x00007f422adca000)
	libmc_solver.so.1 => /usr/local/lib/libmc_solver.so.1 (0x00007f422ab6a000)
	libmc_rtc_gui.so.1 => /usr/local/lib/libmc_rtc_gui.so.1 (0x00007f422a963000)
	libmc_rbdyn.so.1 => /usr/local/lib/libmc_rbdyn.so.1 (0x00007f422a674000)
	libmc_rtc_loader.so.1 => /usr/local/lib/libmc_rtc_loader.so.1 (0x00007f422a469000)
	libmc_rtc_utils.so.1 => /usr/local/lib/libmc_rtc_utils.so.1 (0x00007f422a21b000)
	libyaml-cpp.so.0.6 => /usr/local/lib/libyaml-cpp.so.0.6 (0x00007f4229fa0000)
	libTasks.so.0 => /usr/local/lib/libTasks.so.0 (0x00007f4229ceb000)
	libeigen-lssol.so.1 => /usr/local/lib/libeigen-lssol.so.1 (0x00007f4229aa1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4229798000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f422957b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f422c6a9000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007f4229363000)
	libltdl.so.7 => /usr/lib/x86_64-linux-gnu/libltdl.so.7 (0x00007f4229159000)
	libmc_rtc_ros.so.1 => /usr/local/lib/libmc_rtc_ros.so.1 (0x00007f4228f3e000)
	libmc_observers.so.1 => /usr/local/lib/libmc_observers.so.1 (0x00007f4228d39000)
	libnanomsg.so.5 => /usr/local/lib/libnanomsg.so.5 (0x00007f4228af6000)
	libroscpp.so => /opt/ros/kinetic/lib/libroscpp.so (0x00007f422875a000)
	libroscpp_serialization.so => /opt/ros/kinetic/lib/libroscpp_serialization.so (0x00007f4228557000)
	libtinyxml2.so.2 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.2 (0x00007f4228344000)
	libRBDyn.so.1 => /usr/local/lib/libRBDyn.so.1 (0x00007f42280c3000)
	libsch-core.so.1.0.0 => /usr/local/lib/libsch-core.so.1.0.0 (0x00007f4227e03000)
	libmc_trajectory.so.1 => /usr/local/lib/libmc_trajectory.so.1 (0x00007f4227be0000)
	librostime.so => /opt/ros/kinetic/lib/librostime.so (0x00007f42279b3000)
	libtf2_ros.so => /opt/ros/kinetic/lib/libtf2_ros.so (0x00007f4227708000)
	libtf2.so => /opt/ros/kinetic/lib/libtf2.so (0x00007f42274d4000)
	libmc_rbdyn_urdf.so.1 => /usr/local/lib/libmc_rbdyn_urdf.so.1 (0x00007f42272bf000)
	libgeos-3.5.0.so => /usr/lib/x86_64-linux-gnu/libgeos-3.5.0.so (0x00007f4226f25000)
	libyaml-cpp.so.0.5 => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5 (0x00007f4226ca8000)
	libeigen-qld.so.1 => /usr/local/lib/libeigen-qld.so.1 (0x00007f4226aa1000)
	libboost_timer.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_timer.so.1.58.0 (0x00007f422689c000)
	libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007f4226571000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f422636d000)
	libanl.so.1 => /lib/x86_64-linux-gnu/libanl.so.1 (0x00007f4226169000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4225f61000)
	librosconsole.so => /opt/ros/kinetic/lib/librosconsole.so (0x00007f4225d2f000)
	libxmlrpcpp.so => /opt/ros/kinetic/lib/libxmlrpcpp.so (0x00007f4225b10000)
	libcpp_common.so => /opt/ros/kinetic/lib/libcpp_common.so (0x00007f4225908000)
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007f42256e2000)
	libboost_chrono.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.58.0 (0x00007f42254da000)
	libboost_serialization.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.58.0 (0x00007f4225279000)
	libactionlib.so => /opt/ros/kinetic/lib/libactionlib.so (0x00007f4225059000)
	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007f4224e54000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f4224c15000)
	librosconsole_log4cxx.so => /opt/ros/kinetic/lib/librosconsole_log4cxx.so (0x00007f42249fc000)
	librosconsole_backend_interface.so => /opt/ros/kinetic/lib/librosconsole_backend_interface.so (0x00007f42247fa000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007f4224423000)
	libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007f422411b000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007f4223ee9000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007f4223cc2000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007f4223860000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007f42234cc000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f42232c7000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f422308f000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f4222e66000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007f42213af000)
undefined symbol: _ZTVN5copra20TrajectoryConstraintE	(/usr/local/lib/liblipm_walking_controller.so)
[... cut ...]
undefined symbol: _ZN5copra14TrajectoryCost8autoSpanEv	(/usr/local/lib/liblipm_walking_controller.so)
undefined symbol: _ZN5copra16EqIneqConstraintC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb	(/usr/local/lib/liblipm_walking_controller.so)
undefined symbol: _ZN12lipm_walking10ControllerC1ESt10shared_ptrIN8mc_rbdyn11RobotModuleEEdRKN6mc_rtc13ConfigurationE	(/usr/local/lib/mc_controller/lipm_walking_controller.so)
```
However, I just installed the [newest version](https://github.com/vsamy/copra) and tested both master branch and topic/modern_cmake branch.

What am I doing wrong?

stephane-caron, 2019-11-17:
@ndehio when you mention "versions", it is best if you refer directly to the commit number.

### 1. Compilation troubleshoot

The latest commit of copra is ``22c9dfce60ab5c0017abf7f83fd992c34665cfe7``. That's the one I'm using right now and I have no linking problem. Maybe the controller was not recompiled properly?

- Are you sure lipm_walking_controller is present (copied or linked) from your ``catkin_ws/src``?
- Try deleting ``build/``, ``devel/`` and ``install/`` from your catkin_ws, then do ``catkin_make`` (possibly many times to generate messages) and ``catkin_make install`` again

### 2. Checking that you enable LIPMWalking only

@ndehio From your first error message it looks like mc_rtc is trying to load FogaleController, not the LIPMWalking controller. Did you set ``"Enabled": ["LIPMWalking"]`` in your mc_rtc configuration?

arntanguy, 2019-11-17:
On a side note, I am preparing a catkin-free version of the CMakeLists. I don't really get why this was a catkin package in the first place, and I find it very messy and annoying to deal with ;)

Regarding the problem at hand, my best guess is that you have a mix of libraries between previous installations and that one (especially considering that you've been using custom version of the LIPMWalking). Also make sure that copra is up-to-date (including the cmake submodules).

ndehio, 2019-11-18:
thanks for your help, works for me again =)

closed 2019-11-18 by ndehio