stephane-caron/lipm_walking_controller #30: Wrench distribution QP computation times

stephane-caron, 2019-11-20:
We've been comparing the performance of LSSOL with QuadProg in the ``topic/NoLSSOL`` branch for the wrench distribution quadratic program (WDQP) of the stabilizer. Currently QuadProg solves it much faster:

[image]

On average QuadProg runs around 0.02 ms, vers 025 ms for LSSOL.

stephane-caron, 2019-11-20:
There is a short walking phase (``warmup`` plan) in the plot above, but it is not so clear. Here is a walking plan (``forward_30cm_steps``):

[image]

The difference between DSP and SSP problem solving times is quite clear.

stephane-caron, 2019-11-20:
@aescande Computation times are the same with and without custom QR decomposition for QuadProg:

[image]

At the last DSP of this example, the "no QR" (_i.e._ ``Q = A.transpose() * A``) solution is roughly 2x closer (than the "QR" one) to the one found by LSSOL:

[image]

aescande, 2019-11-21:
In the comparison QuadProgQR vs QuadProdNoQR, are you measuring the time spent to form Q for the NoQR case ?

How come that the earlier timing you did between the controllers based on LSSOL and QuadProgNoQR were very close ?

stephane-caron, 2019-11-21:
> In the comparison QuadProgQR vs QuadProdNoQR, are you measuring the time spent to form Q for the NoQR case ?

Yes, you can check the instructions between ``startTime`` and ``endTime`` in both case the [QR](https://github.com/stephane-caron/lipm_walking_controller/blob/262d4dc76a65860af9d5f8d246433bb2ccea1e46/src/Stabilizer.cpp#L838) and [NoQR](https://github.com/stephane-caron/lipm_walking_controller/blob/262d4dc76a65860af9d5f8d246433bb2ccea1e46/src/Stabilizer.cpp#L830) cases. Actually, the NoQR case has a slight disadvantage here as it also includes the computation of ``c = -A^T b`` :wink: 

> How come that the earlier timing you did between the controllers based on LSSOL and QuadProgNoQR were very close ?

I didn't commit this intermediate version so I can't tell you for sure. In the commit history timings start after we switched to the QR decomposition.

stephane-caron, 2019-11-21:
> Actually, the NoQR case has a slight disadvantage here as it also includes the computation of c = -A^T b

Removing this bias the NoQR version is actually faster:

[image]

stephane-caron, 2019-11-21:
We investigated the performance of LSSOL deeper with @aescande, using settings from [Tasks](https://github.com/jrl-umi3218/Tasks/) to improve its solve times, notably:
```c++
  lssol_.warm(true);
  lssol_.persistance(true);
  lssol_.forceMinSize(false);
```
This brought the number of iterations from 10-40 to 1-4, so that the average solve time drops from 200-400 µs to 50 µs. Incidentally, these are roughly the same solve times than Tasks' IK. Still, much slower than QuadProg:

[image]

Roughly, QuadProg is 6x faster than LSSOL on the double-support QP and 20x faster on the single-support QP.

PR #31 (2019-11-21): Remove dependency on eigen-lssol
Wrench distribution QP is now solved with [QuadProg](https://github.com/jrl-umi3218/eigen-quadprog/). Closes #30.
closed 2019-11-21 by stephane-caron