stephane-caron/lipm_walking_controller #32: Refactor orientation tasks

stephane-caron, 2019-11-29:
The current implementation has ``Controller`` update the two orientation tasks as follows:
```
    sva::PTransformd X_0_a = floatingBaseObs_.getAnchorFrame(controlRobot());
    pelvisOrientation_ = X_0_a.rotation();
    pelvisTask->orientation(pelvisOrientation_);
    torsoTask->orientation(mc_rbdyn::rpyToMat({0, torsoPitch_, 0}) * pelvisOrientation_);
```
This makes it look like orientation tasks are optional, which would be misleading. Here ``X_0_a`` is computed from control ankle frames, which [comply with the ground](https://www.youtube.com/watch?v=MxSJtEKkZE0). If the second line were ``pelvisOrientation_ = Eigen::Matrix3d::Identity();`` the upper body would sway as ankle angles propagate uncompensated to the hip.

I will refactor the code to make this compensation behavior explicit, making orientation tasks part of the stabilizer and adding the corresponding assumption to the spec.

arntanguy, 2019-11-29:
I can confirm that this is problematic for lateral sway in my integration of your work in mc_rtc. I'll keep you posted when I have re-tested this on HRP2.

stephane-caron, 2020-04-25:
This discussion is about a refactoring that wouldn't affect controller performance. It would have been better for a reader's understanding, yet it is now clear I will not do it :wink: So, instead, I have inlined this discussion in-code, and will now close this issue.

closed 2020-04-25 by stephane-caron

stephane-caron, 2020-10-31:
Some more thoughts in a new [wiki page](https://github.com/stephane-caron/lipm_walking_controller/wiki/Why-isn't-there-an-angular-momentum-task-in-whole-body-control%3F) about this question.