stephane-caron/lipm_walking_controller #33: Use of state estimation in the experiments

Ahmadreza-Shahrokhshahi, 2019-12-06:
Hi Dr. Caron,

Based on what is shown in the block diagram in your [paper](https://hal.archives-ouvertes.fr/hal-01875387/document), an state estimation is performed in the "DCM observer" block. However, it seems that in the "LIPM Walking Controller" package, there is no code or consideration for that block (or at least I couldn't find).
Did you use any kind of state estimation/observation methods in the experimental tests?
If so, could you please explain a little bit about it; and if possible, provide references showing the method and details of how you did that?

Thanks a lot

stephane-caron, 2019-12-11:
We sure did, real robots don't walk well without some kind of feedback on state estimates. The DCM observer is not explicitly factored in the code, you will find it in the following components:

- The [FloatingBaseObserver](https://scaron.info/doc/lipm_walking_controller/structlipm__walking_1_1_floating_base_observer.html) does most of the job. It relies on [IMU with anchor frame estimation](https://scaron.info/teaching/floating-base-estimation.html), which is the _de facto_ standard nowadays.
- In [Controller::updateRealFromKinematics](https://scaron.info/doc/lipm_walking_controller/structlipm__walking_1_1_controller.html#af903c7ae95e53fbd687a218e52f91eb6) the controller uses it to compute the CoM position and its velocity.
- These are finally forwarded to [Stabilizer::updateState](https://scaron.info/doc/lipm_walking_controller/structlipm__walking_1_1_stabilizer.html#a98b31d139feb6dc70b0b8c73a8b4437a).

Hoping this helps.

stephane-caron, 2019-12-11:
Closing the issue here, feel free to re-open if you have further questions after checking the above references.

closed 2019-12-11 by stephane-caron

Ahmadreza-Shahrokhshahi, 2019-12-11:
> We sure did, real robots don't walk well without some kind of feedback on state estimates. The DCM observer is not explicitly factored in the code, you will find it in the following components: [...]

I'll try. Thanks a lot.