stephane-caron/lipm_walking_controller #35: Building against mc_rtc 1.0

arntanguy, 2019-12-13:
You should probably consider building against the upcoming `v1.0` release of `mc_rtc` ;)
Very minor changes, basically the only thing to do is replace `GUIState.h` by `<mc_rtc/gui.h>` (and bump your required commit references in your documentation).

I've just merged your `v1.6` release, and made these changes, along with removing the unnecessary ros dependencies in https://github.com/jrl-umi3218/lipm_walking_controller.

Feel free to merge these changes to your own repo if you are fine with a `catkin_free` build. This may require some minor changes to your docker file.

stephane-caron, 2019-12-24:
Sure, let's keep this issue open until mc_rtc 1.0's official release (as a :gift: under the Christmas tree tomorrow? :smile: :santa:)

stephane-caron, 2020-04-25:
Closing this issue now. This repo is for the archival version of the controller, that is, the one that matches the [Docker image](https://hub.docker.com/r/stephanecaron/lipm_walking_controller). Developers who want to install from source and use an up-to-date version of mc_rtc should go for the maintained fork at https://github.com/jrl-umi3218/lipm_walking_controller :wink:

closed 2020-04-25 by stephane-caron