stephane-caron/lipm_walking_controller #36: MPC-tutorial-question: Weight tuning problem or something else?

wyqsnddd, 2019-12-14:
Hi Stephane! 

I managed the first trial run with the weights: 
```json
{
  "initial_plan": "custom_forward",
  "mpc":
  {
    "weights":
    {
      "jerk": 1.0, // Minimizing the input
      "vel": [300.0, 300.0], // CoM-Vel tracking
      "zmp": 1000.0 // ZMP-Pos tracking
    }
  },
```

However it immediately fell down: 

[image]

I am about to debug it tomorrow. But do you think it's a tuning problem or does it more look like a bug inside my MPC?

wyqsnddd, 2019-12-14:
I found a weird plot below: 
[image]

I can't get why the COM velocity generated by the MPC blown off despite the Pendulum COM-Vel reference looks OK.

stephane-caron, 2019-12-24:
For the context, this question was related to the following [walking pattern generation (by model predictive control) tutorial](https://scaron.info/doc/lipm_walking_controller/walkgen.html) where we replace the CoM jerk input in the controller's MPC by the ZMP velocity.

The root of your problem was that, when the MPC formulation changes, you need to update the corresponding [Preview integrator](https://scaron.info/doc/lipm_walking_controller/structlipm__walking_1_1_preview.html) from CoM jerk to ZMP velocity as well. Now that you have done that your solution works fine.

closed 2019-12-24 by stephane-caron

wyqsnddd, 2019-12-26:
Yes, it was an initialization error. Now everything works.