stephane-caron/lipm_walking_controller #37: mc_rtc v1.0 and problem with launch file

xeroxx, 2020-01-10:
Hi @stephane-caron,
I followed the instructions that you provided [here](https://scaron.info/doc/lipm_walking_controller/build.html) to install the lipm_walking_controller from the source. For mc_rtc related packages, I cloned the [recent versions](https://github.com/jrl-umi3218/mc_rtc) from publicly available GitHub repositories.
I've done some modifications based on issue #35 by @arntanguy's comment.
As there is no mc_scenario in the latest mc_rtc_data, how can I use ROS launch file to run the simulation?

arntanguy, 2020-01-10:
Hi,

mc_scenario has indeed be removed from mc_rtc master, the information that it used to provide is now published by mc_rtc itself. The main difference on your side is that you should run (no need to provide the robot explicitely anymore)
```
roslaunch lipm_walking_controller display.launch
```
instead of 
```
roslaunch lipm_walking_controller display.launch robot:=jvrc1
```

xeroxx, 2020-01-10:
> Hi,
>
> mc_scenario has indeed be removed from mc_rtc master, the information that it used to provide is now published by mc_rtc itself. The main difference on your side is that you should run (no need [...]

Thanks @arntanguy for your prompt reply. I checked out to previous commits to see if I can work with them. I'll try your suggestion.

arntanguy, 2020-01-10:
Btw, if you're building against the latest mc_rtc from source, I suggest you use our officially maintained fork https://github.com/jrl-umi3218/lipm_walking_controller.

xeroxx, 2020-01-11:
> Btw, if you're building against the latest mc_rtc from source, I suggest you use our officially maintained fork https://github.com/jrl-umi3218/lipm_walking_controller.

But unfortunately there is no launch file in [that](https://github.com/jrl-umi3218/lipm_walking_controller) fork.

stephane-caron, 2020-04-25:
Closing this issue now. To build from source using more recent versions of mc_rtc I also recommend going for the maintained fork at https://github.com/jrl-umi3218/lipm_walking_controller. This repository will stay as is, with instructions consistent with the software released in the [Docker image](https://hub.docker.com/r/stephanecaron/lipm_walking_controller).

closed 2020-04-25 by stephane-caron