stephane-caron/lipm_walking_controller #38: Problem loading JVRC1 in RViz (using docker)

xeroxx, 2020-01-10:
I run this command in the [docker](https://hub.docker.com/r/stephanecaron/lipm_walking_controller) to run RViz using ROS launch:
roslaunch lipm_walking_controller display.launch robot:=jvrc1
Unfortunately, it seems that the robot model is not loaded properly. Attached is the log file.

[log.txt](https://github.com/stephane-caron/lipm_walking_controller/files/4043593/log.txt)

xeroxx, 2020-01-10:
I just found that lipm_walking is a bash script on the docker (/usr/local/bin/lipm_walking). Based on its contents, I understood that choreonoid simulation should be started before running the roslaunch command.

closed 2020-01-10 by xeroxx

stephane-caron, 2020-01-12:
Hi @xeroxx, yes exactly. For starters, we expect users to run the Docker using the one-liner from the README:
```bash
docker run -it --rm --user ayumi -e DISPLAY=${DISPLAY} -v /tmp/.X11-unix:/tmp/.X11-unix:rw stephanecaron/lipm_walking_controller lipm_walking --floor
```
The process to run Choreonoid and RViz separately is [also documented](https://scaron.info/doc/lipm_walking_controller/docker.html#docker_separate). When you start RViz the robot mesh will be bundled and white like you showed, then when the controller is started the robot description gets published and RViz is updated accordingly.

I updated the doc from your feedback. Thanks!