stephane-caron/lipm_walking_controller #44: CoM Admittance Control

Saeed-Mansouri, 2020-03-13:
Hi Dr. Caron,

1- I have found that you implemented a different approach than what you proposed in the [article](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8794348) for CoM Admittance Control (ZMP compliance control). In the article, you directly modified CoM acceleration by multiplying ZMP error by the CoM admittance gains. However, in the LIPM package, you calculated CoM velocity based on ZMP error. Would you please explain why you considered a different approach?
```
Eigen::Vector3d newVel = -R_c_0 * comAdmittance.cwiseProduct(R_0_c * zmpccError_);
Eigen::Vector3d newAccel = (newVel - zmpccCoMVel_) / dt_;
```
2- I have found that CoM admittance gains are zero by default. It means that CoM Admittance Control doesn't have been used practically. Did you set non-zero values for CoM admittance gains in the experimental tests that you performed on HRP-4 humanoid robot?

stephane-caron, 2020-03-14:
Hi @Saeed-Mansouri,

### Q1: acceleration or velocity control law?

If you look at the history of the repository you will see that I experimented with several variations of this law :wink: In the code from September 2018 ([this experiment](https://www.youtube.com/watch?v=vFCFKAunsYM)) you will see that it was called ``updateCoMAccelZMPCC()`` and purely acceleration-based (no integrator), and that's how I reported it in the [v1 of the manuscript](https://hal.archives-ouvertes.fr/hal-01875387v1/document). Then I added the integrator and found it helpful.

I've kept on updating the manuscript on the way (it is now a [post-print](https://hal.archives-ouvertes.fr/hal-01875387v1/document), *i.e.* the online version is more up-to-date than the proceedings of the conference) but there are still discrepancies between words and code. Thank you for pointing out this one :slightly_smiling_face:

I've updated the post-print to fix this and sent it to HAL. It will be online as soon as they review it. On a side note, to convert from acceleration- to velocity-based admittances I simply multiply by ``dt``, which is 5 [ms] on HRP-4.

### Q2: were you using CoM admittance control in your experiments?

Yes we were (as reported in the paper!) We have used CoM admittance control (ZMPCC) for stair climbing and lateral stepping with HRP-4. If you check the gains at the bottom of the [mobile ground example](https://www.youtube.com/watch?v=MxSJtEKkZE0) you will see that it was also active (although in this example I believe it was only marginally helpful; I didn't check without it though). I also use an extended version of it in the [VHIP stabilizer](https://github.com/stephane-caron/vhip_walking_controller).

Actually, the question you should ask is not "was it used?" but "should I use it?" And the answer to this question depends on your use case. I just wrote my best shot at it in [this comment](https://github.com/jrl-umi3218/mc_rtc/pull/16#issuecomment-599046446).

stephane-caron, 2020-03-14:
Summarized the reason why CoM admittance control is disabled by default in [this wiki page](https://github.com/stephane-caron/lipm_walking_controller/wiki/Pros-and-cons-of-CoM-admittance-control).

Thanks for your feedback!

closed 2020-03-14 by stephane-caron