stephane-caron/lipm_walking_controller #45: slippage of the real-robot during DSP

Saeed-Mansouri, 2020-03-17:
Hi Dr. Caron,

We are using `$ rosrun mc_rtc_ticker mc_rtc_ticker` to connect to the robot. **Assuming the condition that we don't use sensor data (joint-encoder, FT sensor, IMU sensor) as the feedback signal**. As shown in the attached figure, in the RViZ environment we can see that:

1. joint angles of the real-robot are always in the home configuration
2. control-robot tracks the desired walking trajectory
3. real-robot doesn't move during SSP
4. real-robot slips on the ground, as soon as DSP starts
5. during SSP, one of the legs of the real-robot is coinciding with the stance foot of the control-robot

Would you please explain why we should expect that the real-robot slips on the ground during DSP?

start of the DSP:

[image]

end of the DSP:

[image]

Saeed-Mansouri, 2020-03-17:
It has been solved! :)

MohamedNor2020, 2020-03-18:
Hello saeed , execuse me for asking this out of your subject question , how to you import your robot urdf and make it comaptible with the controller because I can't find tutorials for that

stephane-caron, 2020-03-21:
OK great. Can you share the cause of this behavior and how you solved it? It may help other developers who encounter the same problem in the future (avoiding the [DenverCoder9 effect](https://xkcd.com/979/)).

Saeed-Mansouri, 2020-03-27:
I have found that in `void FloatingBaseObserver::estimatePosition(const mc_rbdyn::Robot & realRobot)`, the anchor frame of the control-robot is used to estimate the floating-base position. Based on this method, if the origin of the real anchor frame doesn't coincide with the control anchor frame, the real-robot slips on the ground in the RViZ environment.

closed 2020-04-18 by stephane-caron