stephane-caron/lipm_walking_controller #47: The mc_panel in ROS Rviz flashed and does not work.

kx-guan, 2020-04-25:
Hi, Dear Dr. Caron:
 Thank you very much for discussion every time.
 
 I compiled the new version (1.6) of LIPM stabilizer, with new version (v1.2.1) of mc_rtc
(but replace GUIState.h by <mc_rtc/gui.h> as mentioned in #35) using Ubuntu 16.04LTS.

 When I tried to run simulation in V-REP, the mc_panel in ROS Rviz flashed (appear and disappear about once per second).
 The marker in left side robot flashed with mc_panel, too.
I tried to push the buttons in the mc_panel, but the buttons did not work.
 Last year, the previous version of LIPM stabilizer with the previous version of mc_rtc did not have such phenomenon (using Ubuntu 14.04LTS).
 Could you give me any advice?
  Thank you very much.
      Best regards.

stephane-caron, 2020-04-25:
OK, well sorry to redirect you again but this issue is not related to the code of the LIPM walking controller, but rather to one of its dependencies (the GUI of mc_rtc). I advise you ask the corresponding experts in the [mc_rtc issue tracker](https://github.com/jrl-umi3218/mc_rtc/issues).

kx-guan, 2020-04-25:
OK. I have opened an issue in the mc_rtc tracker.
Thank you.

closed 2020-04-25 by stephane-caron