stephane-caron/lipm_walking_controller #48: Steps required for changing the simulation timestep in choreonoid

mohamadmahdavian, 2020-05-09:
Hello @stephane-caron 

I am trying to change the simulation timestep in choreonoid walking with LIPM. I would like to know what changes are required for changing the timestep of simulation. It is 0.005 sec by default and I need to change it to 0.01. I have encountered some issues and I asked them in mc_rtc package page. But my current question is about the all the steps that I need to take to correctly change the Timestep for the simulation and see it in mc-log file.

Thank You

stephane-caron, 2020-05-15:
Assuming both your simulation and mc_rtc settings are updated properly, as we discuss [there](https://github.com/jrl-umi3218/mc_rtc/issues/35), you will likely have to lower stabilizer gains as a next step. The default gains are suitable for a 200 Hz control frequency. They were chosen not too close to instability, so they might still work with lower frequencies, e.g. 175 Hz, but they are likely to be too aggressive for 100 Hz.

There is a tutorial in the documentation on [tuning stabilizer gains](https://scaron.info/doc/lipm_walking_controller/stabilizer.html).

stephane-caron, 2020-05-16:
Closed by https://github.com/jrl-umi3218/mc_rtc/issues/35#issuecomment-629539949 :wink:

closed 2020-05-16 by stephane-caron