stephane-caron/lipm_walking_controller #49: What is the maximum time-delay in the experimental tests?

Saeed-Mansouri, 2020-05-15:
Hi Dr. Caron,

Regarding the implementation of the LIPM package in the robot,  my question is that what was the data transmission delay in your experimental tests? and what is the maximum permissible time-delay in the control system?

stephane-caron, 2020-05-16:
I don't have answers to these. We did [run some tests using iperf](https://github.com/esnet/iperf/issues/842) but I haven't kept latency data on record. Maybe the experts at [mc_rtc](https://github.com/jrl-umi3218/mc_rtc/) can provide you fresher data.

The same goes for control time delays. This one depends on the gains you are using, so you will want to evaluate it directly on your system using your current gains (rather than relying on data we may have collected a while ago on different robots and for different gains).

Saeed-Mansouri, 2020-05-19:
What we meant by time-delay in the control system is the data transmission delay in the whole system. In our experimental tests, there is about 50ms delay between the control-robot command and encoder data received. we guess this might not be acceptable in walking with high speed.

stephane-caron, 2020-05-21:
Okay, that's pretty large indeed. On HRP-4 this delay is less than 5 ms.

Saeed-Mansouri, 2020-08-11:
Following our previous discussion, I am going to ask my question again in the other way to make it clarify to myself.
The following graph is the knee joint command and knee joint encoder of the robot. This data is recorded in the ROS. It can be seen that there is about 30 msec latency between the command and sensor data (This latency is quite the same for the other joints). My question is that How much is this latency in HRP4? Is this value (30 msec) reasonable? I want to make sure that we are on the right path.

[image]

stephane-caron, 2020-08-29:
This latency would be around 5 ms for HRP-4. Unfortunately I don't have experimental knowledge to share about the effect of different latency values. Your value of 30 ms could very well work out to make an adult-size biped walk, but you will have a lower force-control bandwidth at the ankles (_i.e._ lower ZMP tracking performance) than what we reported for HRP-4.

This might not be a deal breaker, though. For instance Villa _et al._ (2019) showed that we can make a 76 kg adult-size biped robot walk with [MPC sampling periods up to hundreds of milliseconds](https://arxiv.org/pdf/1907.01805.pdf). Sampling period isn't the same thing as latency, but this shows that snappy ZMP tracking is not such a strong requirement for bipedal locomotion.

closed 2020-10-04 by stephane-caron