stephane-caron/lipm_walking_controller #54: zmp weight in MPC

Saeed-Mansouri, 2020-10-01:
Hi Dr. Caron,
To solve the MPC problem in the LIPM package, you consider a zmp weight for both x- and y- directions. I have found that for walking with low speed, the zmp weight in the y-direction (lateral plane) should be low. The LIPM package would be improved if you consider two different zmp weights for the local x- and y-directions (sagittal and lateral planes).

stephane-caron, 2020-10-04:
That's a good idea. Note that the sagittal and lateral directions here should be in a local frame, similarly to what we do for the velocity weights.

In this repo we are keeping the code in the state it ran in the 2019 demos, but feel free to open a PR for the [active fork](https://github.com/jrl-umi3218/lipm_walking_controller). I'll be also around there for review.

stephane-caron, 2020-11-01:
Closing this issue here since this repository is for the archival version of the controller. Feel free to fork it and enhance it in your own version, or even better, to contribute to the active fork so that all improvements are shared with the community :wink:

closed 2020-11-01 by stephane-caron