stephane-caron/lipm_walking_controller #55: measured CoP vs ZMP

Saeed-Mansouri, 2020-10-01:
Hi Dr. Caron,

As you can see in the attached file, the robot is stable based on the measured CoP (represented in the local frame of the support foot) at the beginning of the single support phase (considering half_width = 5 cm). However, based on the measured ZMP, the robot would not be stable at that time. I think that based on the CoP, the measured ZMP should be in the support polygon. It doesn't make sense to me.

[image]

Saeed-Mansouri, 2020-10-02:
I have found that because of the deformation of the robot, the swing foot leaves the ground later than we expects. So the start time of the SSP does not match with the contactstate time that we have in the logs file. I think that the LIPM packge would be improved if you modify the support bounds in the logs file.

[image]

stephane-caron, 2020-10-04:
Be mindful with:

> the robot is stable based on the measured CoP

Stability is a long-term property of the system (actually it is often more accurate for us to use [viability](https://hal.inria.fr/inria-00390555/document)), not an instantaneous measurement. What you are referring to here is [contact stability](https://scaron.info/teaching/contact-stability.html), i.e., as long as the CoP is within the foot area, the planar surface contact should not break/start tilting.

> So the start time of the SSP does not match with the contact state time that we have in the logs file. I think that the LIPM package would be improved if you modify the support bounds in the logs file.

I see your point: plotting the measured DSP/SSP phases (as opposed to the ones from the FSM states right now) would be better for checking the ZMP position with respect to its actual support area. Feel free to open a PR in the [active fork](https://github.com/jrl-umi3218/lipm_walking_controller) to plot the `measured_support_x/ymin/max`, for instance. 

It was a design choice of this controller to base FSM transitions on the planned SSP/DSP durations only. This means, as you observe here, that the robot is usually still in DSP at the beginning of an SSP phase, and will stay in it for a while until the swing foot task has lifted the foot far enough from the ground. You can try variants where FSM transitions depend on measured contact forces as well, but it will likely take some other changes to make this work reliably.

stephane-caron, 2020-11-01:
Closing this issue here since this repository is for the archival version of the controller. Feel free to fork it and enhance it in your own version, or even better, to contribute to the active fork so that all improvements are shared with the community :wink:

closed 2020-11-01 by stephane-caron