stephane-caron/lipm_walking_controller #59: Angular Momentum

Saeed-Mansouri, 2020-10-30:
Dear Dr. Caron,

You have considered the linear inverted pendulum model for the walking pattern generation. The comtask used for solving inverse kinematics can guarantee that the com heigh of the controlrobot would be constant. I couldn't find any task/strategy that maintains the contolrobot moves with a constant angular momentum.

stephane-caron, 2020-10-31:
This is a very good question :slightly_smiling_face: I started answering you here, than the answer grew in size and I moved it to a [wiki page](https://github.com/stephane-caron/lipm_walking_controller/wiki/Why-isn't-there-an-angular-momentum-task-in-whole-body-control%3F).

Feel free to close this issue if this answers your question, or continue the discussion here otherwise.

Saeed-Mansouri, 2020-11-02:
It was so helpful. Thanks.

closed 2020-11-02 by Saeed-Mansouri