stephane-caron/lipm_walking_controller #61: Stop during the first DSP

Saeed-Mansouri, 2020-12-21:
Hi Dr. Caron

The below command applied in `ModelPredictiveControl::computeZMPRef()` might be activated not only for stopping during the first DSP. I think that you should replace the total steps of the target support with `nbTargetSupportSteps_`.

`if(nbTargetSupportSteps_ < 1) // stop during first DSP`

stephane-caron, 2020-12-29:
This is indeed a controller-specific quirk: the function has two modes (full preview and half preview, documented in [phaseDurations](https://scaron.info/doc/lipm_walking_controller/structlipm__walking_1_1_model_predictive_control.html#abc0a5abba15c0c3c3ff9b0255d4c8138)), and it is only called from `SingleSupport.cpp` and `DoubleSupport.cpp` with either:
- `targetSupportDuration == 0.`: stop during the first DSP, i.e. the DSP between `initContact` and `targetContact`
- `targetSupportDuration == ctl.singleSupportDuration()`: keep going
So, with this implementation the condition will only be triggered for stopping during the first DSP. What's bad about this code is that it is not apparent from the `ModelPredictiveControl` class itself.

I added some doc on this in 3c73788f206309485a5cb64f37fb3dda4fee15de. Also, taking a second look at this piece I realize that `phaseDurations` is not meant to be called with `initSupportDuration + doubleSupportDuration` greater than the total duration (in that case, `targetSupportDuration` would be zero although the intent is not to stop at the first DSP). I added a note on this as well.

stephane-caron, 2020-12-29:
@Saeed-Mansouri Thank you for your feedback :grinning: 

I'll be closing the Issue tracker of this repository since bugs and fixes should go to the [tracker of the active fork](https://github.com/jrl-umi3218/lipm_walking_controller/issues). Feel free to use the new [Discussions](https://github.com/stephane-caron/lipm_walking_controller/discussions) tab for general questions and discussions.

closed 2020-12-29 by stephane-caron